riscv/riscv-p-spec #2: Question about overflow reg csr address?

taoj33, 2020-07-24:
Currently, the ”ucode“ reg locates in CSR address 0x801 which may conflict with other customized regs in some implementations. Would it be more appropriate if it can be placed in a standard region similar to the F/D extension or V extension?

chuanhua, 2020-08-07:
v0.8 release has changed the custom ucode CSR in 0x801 to a standard user read/write CSR (vxsat) in 0x009.
Thanks for the question.

taoj33, 2020-08-25:
In V-spec,  VS bits are used to control accessing vxsat as well as other vector-related csrs and also indicate the context switch status for vector registers.  
As vxsat is used,  does it imply VS bits still exist in p-extension and have similar functions?

chuanhua, 2020-08-26:
In P-extension, only vxsat CSR is the additional state that needs to be saved and restored on context switch. So there is no need to do lazy save/restore like vector extension.

My suggestion is that if P-extension is present, then software should always save/restore vxsat CSR.

chuanhua, 2020-08-26:
And the OFF state of VS field should not affect the execution of the P extension instructions.

closed 2020-09-01 by taoj33