riscv/riscv-p-spec #3: Possible errors in spec require clarification

taoj33, 2020-08-20:
HI, 

When checking the spec we find following mismatches/typos in the spec which may require clarification or correction.

- Instructions only exist in the detailed description but not the instruction list:
      swap16/8, msubr32
- Instructions only exist in the encoding table：
      s/umul16h, s(u)mulx16h, mltbi, mtlei
- possible typos in the encoding table
     ksllw,kslliw  are listed as ksll,kslli
     kmmwat2, kmmwat2 are listed as kmmwt2, kmmwt2.u
     kdmabb16, kdmbb16，khmbb16, kmmawb2.u func7 mismatch with their detailed descriptions

chuanhua, 2020-08-26:
Thanks for your feedback. I will check and correct these mismatch/errors.

taoj33, 2020-08-26:
in addition, clo8/16/32 subf5 also mismatch

chuanhua, 2020-09-04:
Thanks. I have found the subf5 mismatch issues. They will be fixed in the next release.

closed 2020-09-14 by taoj33