riscv/riscv-p-spec #5: Would riscv-p-spec provide more background of the instructions

davidmlw, 2020-09-04:
riscv-p-spec seems a huge extension and has some overlaps with other specs.

Is that possible to provide some more background of the instruction, and list the instructions according to background? And will riscv-p-spec release a set of riscv-compliance test cases?

prikryl-codasip, 2020-09-30:
Having a table that classifies the instructions can help with split into smaller groups/optional parts. The current proposal is a huge one, and not everybody needs all instructions. There can be a good reason to disable some parts of the P extension, so the final CPU is smaller yet powerful enough.

chuanhua, 2020-10-20:
Chapter 2 of the spec classifies the instructions into several categories. Do these classifications match your idea? Or you have other thoughts about appropriate classifications?

prikryl-codasip, 2020-10-20:
Hi Chuanhua,

I'd like to see more subsets in chapter 3. So, something similar to what 
B extension has. In other words, B is split into a smaller subset, and a 
user can pick and choose the right one. It seems that the P extension is 
quite monolithic at the moment.

Chapter 2 contains classes of instructions but not the most useful 
subsets. For instance, if we target something like ARM's DSP extension 
in the M series, it's a combination of several classes in chapter 2.

So, are there any plans to refine the subsets into smaller chunks?

Thanks and best regards.