riscv/riscv-p-spec #10: Question about zp64 subset

taoj33, 2020-11-16:
Hi, 

Based on the description in chapter 3， 

> facilitate the trade-off between implementation complexity and application performance

is it a correct interpretation that ZP64 is an optional subset for RV32P?

if so, is there an abbreviation or any other way to distinguish an ISA group with zp64 extension from the standard one?

Geng-Qi, 2020-11-16:
I happen to be confronted with this conundrum.
I believe that RV32 can be used in combination with RV32 + ZP64 without support for P extensions.
However, is it possible to use the combination of RV64 + ZP64 without supporting P extensions?

chuanhua, 2021-02-01:
The intended combination for RV32P is either "Zp-others" or "Zp-others + Zp64" .
The intended combination for RV64P is just "Zp-others + Zp64" since the 64-bit operand is naturally a single register in RV64.
Zp-others is planned to be called "Zpn" in the next release of the P spec.

chuanhua, 2021-09-29:
Zp64 is now renamed to Zpsfoperand.

closed 2021-09-29 by chuanhua