riscv/riscv-p-spec #11: Questions about KMMAWB2/KMMAWB2.u/KMMAWT2/KMMAWT2.u

wzy971210, 2020-11-18:
Hi,
I have some questions about these 4 instructions. 
It seems that we need to do saturation operation twice when executing, once after multiplication and once after addition.

> if ((Rs1.W[x] == 0x80000000) & (Rs2.W[x].H[1] == 0x8000)) {
  addop.W[x] = 0x7fffffff;
  OV = 1;
} else {
  Mres[x][47:0] = Rs1.W[x] s* Rs2.W[x].H[1];
  if (“.u” form) {
    Mres[x][47:14] = Mres[x][47:14] + 1;
  }
  addop.W[x] = Mres[x][46:15]; // doubling
}
res[x] = Rd.W[x] + addop.W[x];
if (res[x] > (2^31)-1) {
  res[x] = (2^31)-1;
  OV = 1;
} else if (res[x] < -2^31) {
  res[x] = -2^31;
  OV = 1;
}
Rd.W[x] = res[x];
for RV32: x=0
for RV64: x=1…0

1. Is that a must to saturate the MSW Q31 multiplication results? I mean, I can preserve the full accuracy of multiplication results in my implementation. Saturation after multiplication will make my codes more complicated instead.
2. If it is a must to saturate, could you tell me the application scenarios of these 4 insts?

Thanks.

chuanhua, 2020-11-23:
The first saturation is only for one special case (the multiplication of maximum negative numbers) and can be detected on the source operands. We defined the instruction this way, instead of preserving the maximum accuracy, to match the basic operation of the reference codes of many voice/speech codecs for bit-exact compliance. For example, the following basic operation is defined in the Enhanced Voice Service codec:

```
void Mpy_32_16_ss( Word32 L_var1, Word16 var2, Word32 *L_varout_h, UWord16 *varout_l) {
   Word16 var1_h;
   UWord16 uvar1_l;
   Word40 L40_var1;
   if( (L_var1 == ( Word32) 0x80000000) && (var2   == ( Word16) 0x8000)) {
      *L_varout_h = 0x7fffffff;
      *varout_l = ( UWord16) 0xffff;
    } else {
      uvar1_l = extract_l( L_var1);
      var1_h = extract_h( L_var1);
      /* Below line can not overflow, so we can use << instead of L40_shl.  */
      L40_var1 = (( Word40) (( Word32) var2 * ( Word32) uvar1_l)) << 1;
      *varout_l = Extract40_L( L40_var1)
      L40_var1 = L40_shr( L40_var1, 16);
      L40_var1 = L40_mac( L40_var1, var2, var1_h);
      *L_varout_h = L_Extract40( L40_var1);
   return;
}

Word32 Madd_32_16(Word32 L_num, Word32 a, Word16 b)
{
    Word32 result;
    UWord16 lo;
    /* use Mpy_32_16_ss(): */
    Mpy_32_16_ss(a, b, &result, &lo);
    result = L_add(L_num, result);
    return result;
}

```

lazyparser, 2021-01-16:
Hi,

Is this issue got resolved or not?

chuanhua, 2021-10-05:
@wzy971210 Can we close this issue now?