riscv/riscv-p-spec #12: Undefined signedness of operations

mkostal-codasip, 2020-11-18:
The specification at the beggining introduces signed and unsigned multiplication.
But only a few of the instruction definitions use specifically signed or unsigned multiplication.
For some cases that is not a big issue, because from the C-function, we can see, whether the operans are signed or unsigned, but for instructions such as MSUBR32, MADDR32 missing the information can lead to different results.

The definition of these instructions states: `The contents of Rs1 and Rs2 can be either signed or unsigned integers.`
In this case the signedness of a number has a mutually exclusive nature, since from the pseodocode definition we can observe, that signedness can produce different results.

Can the devs tell, what is the intended signedness of instructions MSUBR32, MADDR32

lazyparser, 2021-01-16:
Hi,

Is this issue got resolved or not?

chuanhua, 2021-10-05:
@mkostal-codasip Can you give an example for msubr32/maddr32 where the signedness of the source operands can produce different results? We thought that the signedness of the source operands will not affect the end result. We could be wrong. Thanks.