riscv/riscv-p-spec #13: Missing intrinsic functions definition

mkostal-codasip, 2020-11-18:
For the following instruction I am missing intrinsic functions definition in the spec, could the devs update these:

- MSUBR32 

- MULR64

- MULSR64

- MADDR32

chuanhua, 2021-01-06:
The reason for no intrinsic functions for these instructions is that we think a compiler can use these instructions easily. So a programmer does not need to use these intrinsic functions in their C program. If you think this reasoning is not correct, we can still define intrinsic functions for these instructions.

higuoxing, 2021-01-07:
Hi @chuanhua, thanks for your explanation! I think it might be good to mention it in the spec or somewhere?

chuanhua, 2021-01-21:
Thanks for your suggestion. I will add a note about this for these instructions.

mkostal-codasip, 2021-01-27:
With the release 0.9.1 the missing intrinsic functions have been added. [](https://github.com/riscv/riscv-p-spec/commit/976dbe79a948648aefc73dcc7ea8e640c1b7b4cf)
I suggest to clese this issue

marcfedorow, 2021-04-22:
This definitions seems to be wrong, e.g. `int32_t __rv__maddr32(int32_t a, int32_t b);` takes 2 arguments while 3 should be given.

chuanhua, 2021-04-22:
Thanks for finding this issue. It should have 3 arguments. I will fix it in the next release.

chuanhua, 2021-10-05:
This has been fixed in the latest release. Closing this issue.

closed 2021-10-05 by chuanhua