riscv/riscv-p-spec #14: Inconsistencies and typos

mkostal-codasip, 2020-11-18:
I suggest to make uniform the following pseudocodes:

section 5.128 but not limited to it
`if (sa != 0) {` change to `if (sa u> 0) {`

section 5.96, 5.97 but not limited to it
`Rd.B[x] = (Rs1.B[x] {le} Rs2.B[x])? 0xff : 0x0;` change to `Rd.B[x] = (Rs1.B[x] s<= Rs2.B[x])? 0xff : 0x0;` I suggest to use just `s<=` xor `{le}`

section 2.2.5, table 5
operation description uses gray highlighting unlike other instructions, suggest to use same style

section 2.2.9, table 9 and 10 but not limited to it
`rt.H[x] = (ra.H[x] < rb.H[x])? ra.H[x] : rb.H[x];` change to `rt.H[x] = (ra.H[x] s< rb.H[x])? ra.H[x] : rb.H[x];`

section 2.2.9, table 9 and 10
rows 8-10 do not have operation described, similarly table 10

section 2.2.11
`There are 8 instructions here.` should be `There are 10 instructions here.`

section 2.3.3, table 14
missing No. at row 16

somewhere
Some places use `Rs1.W[x].H[y]` and some other `Rs1.W[x][15:0]` I would suggest to use same style everywhere

section 6.8 but not limited to it
`if ( bool_expression | bool_expression2)` should be `if ( bool_expression || bool_expression2)` as single pipe is generally used for bitwise operation and double pipe is used for logical operations

section 2.5.2, table 23, kslraw
operation describtion missing `if`

chuanhua, 2021-02-01:
Thanks for your suggestion. These will be fixed in the next release of the specification.

chuanhua, 2021-10-12:
I have changed `if (sa u> 0) {` back to `if (sa != 0) {` to reduce the confusion of thinking `sa` is a signed number.

chuanhua, 2021-10-12:
@mkostal-codasip Can this issue be closed? Thanks.