riscv/riscv-p-spec #22: Question about pair x0,x1 in zp64

taoj33, 2020-12-03:
Hi，

Is reg pair x0,x1 allowed in RV32 ZP64 subset? 
If so，should it be used as 64-bit zero， or its real-time data value?

chuanhua, 2020-12-04:
The reg pair of x1.x0 is allowed in RV32 ZP64 subset. However, x0 will always be zero. So most of the time, it will not be used unless you have a 64-bit value with 32-bit zero in the lower part. I think a compiler will not use the x1.x0 pair for register allocation. Only 15 reg pairs will be used by the compiler.

taoj33, 2020-12-09:
> The reg pair of x1.x0 is allowed in RV32 ZP64 subset. However, x0 will always be zero. So most of the time, it will not be used unless you have a 64-bit value with 32-bit zero in the lower part. I [...]

what if pairs index start with 2d+1, should they be defined as illegal instructions or used as index start with 2d?

chuanhua, 2020-12-09:
For a register pair, only the pair group index [4:1] is meaningful. The register index[0] will be ignored and its value is a don't-care value.

closed 2020-12-10 by taoj33