riscv/riscv-p-spec #24: Should instructions UMUL8/UMULX8/UMUL16/UMULX16 be taken as ZP64 instrucions?

wzy971210, 2020-12-15:
Hi

Should instructions UMUL8/UMULX8/UMUL16/UMULX16 be taken as ZP64 instrucions?
They product a 64 bits result and write result back to register-pairs.
[https://github.com/riscv/riscv-p-spec/blob/master/P-ext-proposal.adoc#zp64-for-rv32-only](url)

mkostal-codasip, 2020-12-15:
To my understanding Zp64 is meant to extend RV32 with operations on 64bit words, instructions, that you ask about work with only 8bit and 16bit words, so to me it does not make sense to consider them for Zp64.

I am no member of the P-extension task group, so do not take me seriously.

chuanhua, 2020-12-16:
Yes. These instructions should be put into the Zp64 subset. They produce 64-bit result. I will add them in the next release of the spec. Thanks for finding this issue.

wzy971210, 2020-12-17:
Thanks for replying.

I still have some doubts about ZP64 instructions.
1.  Is Zp64 instrunctions is a must for P-extension in RV-32? In other words, can we just implement Zp64 insts without other P insts?  Also, can we just implement the other P insts?
2.  Is it legal to use odd register to represent Rd, like UMUL8 x7,x8,x9? Is it equivalent to UMUL8 x6,x8,x9?

@chuanhua

Geng-Qi, 2020-12-24:
Hi，
1. Following a similar line of thought, do WEXT/WEXTI belong to zp64?
2. Since KADD64/KSUB64/... belongs to zp64, do CLROV and RDOV belong to zp64?

chuanhua, 2021-01-26:
@wzy971210,
1. Zp64 is not a must for P-extension in RV32. The intended combination is either Zp-others or "Zp-others + Zp64".
2. For a register-pair operand, you can use either even or odd register number to represent the pair. An implementation should just use the pair group number to form the two register numbers in the pair.

@Geng-Qi ,
1. WEXT/WEXTI should be put into Zp64. I will add them to the group.
2. Since the intended combination is either Zp-others or "Zp-others + Zp64", there is no need to put CLROV and RDOV into Zp64.