riscv/riscv-p-spec #26: Some issues I found in this doc.

higuoxing, 2021-01-04:
Hi, I found some issues in this doc. It would be great if someone could take a look :-)

#### Some intrinsic functions are missing:

[5.71. MADDR32 (Multiply and Add to 32-Bit Word)](https://github.com/riscv/riscv-p-spec/blob/master/P-ext-proposal.adoc#571-maddr32-multiply-and-add-to-32-bit-word)
[5.74. MSUBR32 (Multiply and Subtract from 32-Bit Word)](https://github.com/riscv/riscv-p-spec/blob/master/P-ext-proposal.adoc#574-msubr32-multiply-and-subtract-from-32-bit-word)
[5.75. MULR64 (Multiply Word Unsigned to 64-bit Data)](https://github.com/riscv/riscv-p-spec/blob/master/P-ext-proposal.adoc#575-mulr64-multiply-word-unsigned-to-64-bit-data)
[5.76. MULSR64 (Multiply Word Signed to 64-bit Data)](https://github.com/riscv/riscv-p-spec/blob/master/P-ext-proposal.adoc#576-mulsr64-multiply-word-signed-to-64-bit-data)

#### Some intrinsic function names doesn't match the instruction name:

[5.54. KSLLIW (Saturating Shift Left Logical Immediate for Word)](https://github.com/riscv/riscv-p-spec/blob/master/P-ext-proposal.adoc#554-kslliw-saturating-shift-left-logical-immediate-for-word):
I think the intrinsic function should be
```
intXLEN_t rvkslliw(intXLEN_t a, uint32_t b);
                ^
```
rather than
```
intXLEN_t rvksllw(intXLEN_t a, uint32_t b);
```

[5.56. KSLLI8 (SIMD 8-bit Saturating Shift Left Logical Immediate)](https://github.com/riscv/riscv-p-spec/blob/master/P-ext-proposal.adoc#556-kslli8-simd-8-bit-saturating-shift-left-logical-immediate):
Should be:
```
uintXLEN_t __rv__kslli8(uintXLEN_t a, uint32_t b);
                     ^
```
Rather than:
```
uintXLEN_t __rv__ksll8(uintXLEN_t a, uint32_t b);
```

[5.58. KSLLI16 (SIMD 16-bit Saturating Shift Left Logical Immediate)](https://github.com/riscv/riscv-p-spec/blob/master/P-ext-proposal.adoc#558-kslli16-simd-16-bit-saturating-shift-left-logical-immediate)
Should be:
```
uintXLEN_t __rv__kslli16(uintXLEN_t a, uint32_t b);
                     ^
```
Rather than:
```
uintXLEN_t __rv__ksll16(uintXLEN_t a, uint32_t b);
```

[5.101. SLLI8 (SIMD 8-bit Shift Left Logical Immediate)](https://github.com/riscv/riscv-p-spec/blob/master/P-ext-proposal.adoc#slli8-simd-8-bit-shift-left-logical-immediate):
Should be:
```
uintXLEN_t __rv__slli8(uintXLEN_t a, uint32_t b);
                    ^
```
Rather than:
```
uintXLEN_t __rv__sll8(uintXLEN_t a, uint32_t b);
```

chuanhua, 2021-01-21:
All of the intrinsic functions of the small immediate type instructions can be the same as the intrinsic functions of the register-type instructions. This is because a compiler can detect constant value (immediate) in the function argument and determine if the constant range fits inside the immediate range. If the constant value fits in the immediate range, the compiler will generate the immediate type of the instruction.

higuoxing, 2021-01-21:
Thanks for the reply!

closed 2021-01-21 by higuoxing