riscv/riscv-p-spec #28: Open discussion for extra DSP instructions extended by Nuclei

fanghuaqi, 2021-01-15:
Hi all, I am Huaqi from Nuclei, in Nuclei we have implemented P-Ext 0.5.4 and extended some instructions to it, and in 2021.1.15, we have a sync meeting with Andes, PLCT, we have discussed maybe Nuclei can share the extended instructions to see whether it can be added to the mainline of P-Ext, and I have opened a email disscussion see https://lists.riscv.org/g/tech-p-ext/message/94, and attached our extra instruction document there. Any feedback is welcomed.

Thanks
Huaqi

chuanhua, 2021-02-02:
It seems that the following 10 instructions (Dxxxxx) in your "Nuclei SIMD Additional Instructions" are the same as the RV64 xxxxx instructions already in the specification. Why can't you use the RV64 xxxxx instructions already in the spec?

DKHM8 <-> KHM8
DKHM16 <-> KHM16
DKABS8 <-> KABS8
DKABS16 <-> KABS16
DKSLRA8 <-> KSLRA8
DKSLRA16 <-> KSLRA16
DKADD8 <-> KADD8
DKADD16 <-> KADD16
DKSUB8 <-> KSUB8
DKSUB16 <-> KSUB16

fanghuaqi, 2021-02-02:
Hi, the new introduced DSP instructions are only available in RISC-V 32bit, which are created to use pair-registers and used in DSP/NN library porting.

chuanhua, 2021-02-02:
Thanks. The document does not mention using pair-registers. It is clear to me now.