riscv/riscv-p-spec #34: duplication of B opcodes

jnk0le, 2021-02-13:
I see that there is push for O&C review and freeze/ratification because "P is mature enough", but the overlap with B ext and resulting opcode duplication seems to not been adressed yet.

Most of the duplications look quite accidental but the BPICK is literal duplication of CMIX and that takes a lot of encoding space in R4 form.

Do I see some weird EEE game or there are plans to move Zbt entirely into OP-P, As the 3rd read port for ternary opcodes on P enabled cores will be almost free, thus much higher likehood to implement Zbt?

rdolbeau, 2021-02-14:
The subject of ternary (R4 or destructive) also appear in a B issue: https://github.com/riscv/riscv-bitmanip/issues/110

Personally, I think R4 takes too much opcode space for most if not all of those instructions, the destructive form (sourcing RS3 from RD) might make more sense. It's extremely likely to not impact the use of `pbsada` (P), `cmix` (B) (a.k.a. `bpick` (P) as you said), `cmov` (B), `*maqa` (P), ... For me the only doubt are the funnel shifts `fsr`, `fsl` - is there some evidence those need the R4 form?

Also it probably makes more sense for the redundant instructions to be part of B rather than P (or be in both using the same opcodes), as P is much more complex to implement - beyond the third read port (only used by Zbt), you also need a second write port for Zp64 in RV32 (and it's quite useful to have Zp64, as e.g. it seems all SIMD multiplications but the specialized `pbsad*` and `*maqa` are in Zp64).

jnk0le, 2021-02-17:
FYI
I have filed proposal to move Zbt into reserved FMA4 encoding riscv/riscv-bitmanip/issues/112

chuanhua, 2021-03-18:
The duplication scenarios between P and B are a little bit complicated except for BPICK (CMIX) instruction which duplicates in both RV32 and RV64. The other duplications happen in either RV32 or in RV64, but not both. The solution depends on how we want to manage this kind of difference. Do we just want to remove the duplication if an instruction is duplicated in both RV32 and RV64? Or do we want to remove the duplication if an instruction is duplicated in either RV32 or RV64? In the latter case, the duplication removal list will be different between RV32 and RV64.

rdolbeau, 2021-03-18:
@chuanhua @jnk0le Indeed there's some 'partial' overlap, shouldn't those be addressed somehow ? First step would be to document them I guess, has that been done already (chapter 4 of P only documents 'full' overlap it seems) ?

In case not, here's some overlaps that I noticed during implementation, though I didn't investigate in details:

`bitrev[i]` has overlap with `grev[i]` (but isn't a strict subset, as `bitrev[i]` manipulates an arbitrary number of bits whereas `grev[i]` is register-wide)

halfword (16 bits) movement instructions appear in both with some overlaps at least in RV32: `swap16` and `pk[bt][bt]16` (also #32) in P, `packh` and `xperm.h` in B, `grev[i]` (e.g. pseudo `rev16`) as well

Similar overlap within the 32 bits instructions.

`swap8` from P is `rev8.h` from B (pseudo from `grevi`)

`clz32`, `clo32`, `clrs32` in P and `clz`, `ctz` in B ; P has the shorter versions but not B - but B has the zero-/sign-extend instructions to more efficiently implement the byte/halfword/word version with the XLEN version.

`wext` from P with `fs[lr]` in B

`insb` from P with `bfp` from B

`bpick` (P) is `cmix` (B) as already mentioned, along with the `min` and `max` stuff (P chapter 4 stuff).

chuanhua, 2021-03-25:
INSB from P and BFP from B does not seem to overlap. It takes 3 instructions by using BFP to emulate INSB, right?

```
Rv32:

insb rd, rs1, uimm2 (P)

addi t0, zero, {0b1000, [0b00000, 0b01000, 0b10000, 0b11000]}
pack t0, a0, t0
bfp t0, a1, t0   (B)

RV64:

insb rd, rs1, uimm3 (P)

lui t0, zero, {3'b100, 0b01000, [0b000000, 0b001000, 0b010000, 0b011000,
                                 0b100000, 0b101000, 0b110000, 0b111000], 4'b0000}
pack t0, a0, t0
bfp t0, a1, t0    (B)

```

rdolbeau, 2021-03-25:
@chuanhua Yes for `bfp` you first need to specify the pattern in a register somehow first, which indeed might be costly.

`insb rd, rs1, uimm2` is `bfp rd, rd, rs4` when `rs4 == 0x080${zext(uimm2,4)}00${rs1[7 downto 0]}`

That makes me think an immediate-form of `bfp` might be more general than `insb` while requiring less setup than the full reg-reg `bfp`.

rdolbeau, 2021-03-31:
@chuanhua Oups my bad; for 16-bits / halfword manipulations are `pack` and `packu` in B (the dot-less h suffix I find confusing).

aswaterman, 2021-05-25:
I am going to file issues for each of the overlap cases, because the resolution might differ for each.

jnk0le, 2021-05-25:
Closing as separated into dedicated issues.