riscv/riscv-p-spec #35: inconsistent reg specifier names

jnk0le, 2021-02-13:
Chapter 2 uses `rt, ra, rb` form which looks like artifact from other architecture, the extension was ported from.
Chapter 5 suddenly switches to `rd, rs1, rs2` form.

There is still a problem with destructive operand naming. `rd` might not be the most convinient, and the crypto group already "invented" `rt` operand as a destructive one, but their `rt` lies in rs1 (that also conflicts with chapter 2 naming).

chuanhua, 2021-03-25:
Chapter 2 is instruction summary with small table cells. So I have the operand name shortened for operation description in the constrainted space.
Chapter 5 is detailed instruction description, so I used the conventional form of register names.

chuanhua, 2021-09-06:
Fixed in 20d76b4.

closed 2021-09-06 by chuanhua