riscv/riscv-p-spec #39: overlap with RVI

jnk0le, 2021-03-17:
`ADD64` and `SUB64` on RV64 are duplicating `ADD` and `SUB` from base instruction set.

ADD64
```
RV64:
rt = ra + rb
```
SUB64
```
RV64:
rt = ra - rb
```

chuanhua, 2021-03-18:
This is true. It is intentional to keep the extension simple to manage between RV32P and RV64P. Another solution is to remove these duplicated instructions from RV64P.

aswaterman, 2021-05-25:
My take is that these should be removed from RV64P.

marcfedorow, 2021-09-09:
"The same name (encoding) for the same thing" is not the RISC-V approach.
In my ([and not only my](https://gist.github.com/erincandescent/8a10eeeea1918ee4f9d9982f7618ef68)) opinion it is rather a disadvantage than a virtue (i.e. now zext.h has two different encodings depending on xlen) but the naming scheme should be followed for consistency.
This means that #61 and #79 should be fixed (otherwise p-ext insns will be hard to read among other risc-v asm).
Thus no such consistency between rv32 and rv64 (as RVI overlap may add) is reachable.
If you still think that it is critical to have add64 insn on rv64, you may want to declare it as an rv64-only alias to add but it obviously should not have its own encoding on rv64.
Remove ADD64 at all and declare ADD.PAIR (rv32-only) is a much better solution tho.

chuanhua, 2021-09-27:
ADD64 and SUB64 are removed from RV64 Zpsfoperand extension in e274869.

closed 2021-09-27 by chuanhua