riscv/riscv-p-spec #42: about the availability of Zpsfoperand subset

linsinan1995, 2021-04-23:
I have some confusion on several instructions under the Zpsfoperand subset. p-spec v0.93 shows that  Zpsfoperand is for RV32 Only in [chapter3](https://github.com/riscv/riscv-p-spec/blob/master/P-ext-proposal.adoc#zpsfoperand-for-rv32-only), but instructions like:
```
smul16
smulx16
umul16
smal
smar64
smsr64
umar64
umsr64
kmar64
kmsr64
ukmar64
ukmsr64
smalbb
smalbt
smaltt
smalda
smalxda
smalds
smaldrs
smalxds
smslxda
wext
```

also has a description of the operation for RV64. Did I misunderstand anything here?

chuanhua, 2021-04-26:
The original thinking about all instructions in Zpsfoperand in RV64 is to categorize them all in Zpn (RV64), so only RV32 has the sub-extension of Zpsfoperand to include all register-pair instructions. However, we later realized that this may cause confusion about the sub-extension of these instructions. So we have decided to make Zpsfoperand a sub-extension in both RV32 and RV64. The next release will remove the "RV32 Only" wordings for Zpsfoperand. Thanks for your question.

linsinan1995, 2021-04-26:
Thanks for answering.

chuanhua, 2021-09-29:
Closing this issue.

closed 2021-09-29 by chuanhua