riscv/riscv-p-spec #43: typos in maddr32 & msubr32?

linsinan1995, 2021-04-26:
according to the desc, 
> Description: This instruction multiplies the lower 32-bit content of Rs1 with that of Rs2, subtracts the lower 32-bit multiplication result from the lower 32-bit content of Rd, then writes the final result (RV32) or sign-extended result (RV64) back to Rd. The contents of Rs1 and Rs2 can be either signed or unsigned integers.

the intrinsic functions of msubr32 sound more like 
```
intXLEN_t __rv__msubr32(int32_t a, int32_t b);
```
instead of 
```
int32_t __rv__msubr32(int32_t a, int32_t b);
```
to me (the same for maddr32). Did I get something wrong?

chuanhua, 2021-04-28:
The description of RV64 is for writing a result with a signed 32-bit range into a 64-bit register. However, the return type of the intrinsic function does not need to match the 64-bit register result that has a signed 32-bit range. Most of the time, using `int32_t` as the return type will let a compiler generate more efficient code than using `int64_t` as the return type. Because if the following use of the returned result is in the type of `int32_t`, no extra instructions will be generated. Otherwise, the `int64_t` type data needs to be processed into a `int32_t` data by a compiler with extra instructions which are unnecessary.

marcfedorow, 2021-04-30:
> let a compiler generate more efficient code
Using [u]int_fast8_t could be more verbose.