riscv/riscv-p-spec #44: Intrinsic function naming scheme for P extension.

kito-cheng, 2021-04-29:
I've create an issue on riscv-c-api-doc: https://github.com/riscv/riscv-c-api-doc/issues/18, which listed all current naming scheme for all intrinsic functions.

I think it would be great if we have consistent intrinsic function naming scheme for all extension.

marcfedorow, 2021-04-29:
+1.
Also I disagree with `__rv__sadd64` and `__rv__ssub64` names.
There are `__rv__kadd64` and `__rv__ukadd64`.
There are `__rv__radd64` and `__rv__uradd64`.
I think we could similarly have `__rv__add64` for signed tydes and `__rv__uadd64` for unsigned.
(The same for subtraction insns).

chuanhua, 2021-09-15:
Release v0.9.6 have changed `__rv__` to `__rv_`.

pz9115, 2026-03-26:
Out of date, mark as closed.