riscv/riscv-p-spec #45: Intrinsic types

marcfedorow, 2021-04-29:
There are more `SEXT32(rd)` functions but they does not follow the `maddr32`'s logic (check #43 for more information). E.g. `kaddh`, `kaddw`, `ksubh`, `ksubw`, `raddw`, `rsubw`and their unsigned analogues return `intXLEN_t` and `uintXLEN_t` respectively. I think that all such intrinsics should return `[u]int32_t`.
Also sign-extension of unsigned results (e.g. in `uraddw`) seems strange to me.

There are more problems with intrinsic types:
I think, `kabsw` may return `uint32_t`; `SE64` may be removed.
`kslra8` and `kslra8.u` should take the second argument as `int32_t`, not `int`. Same for `kslra16[.u]`.
`smul8`, `smulx8`, `umul8`, `umulx8`, `umul16`, `umulx16` should take `uint32_t` arguments as `smul[x]16` does.
`sunpkd8xy` intrinsics should probably return `intXLEN_t` instead of `uintXLEN_t` while it is a signed unpack operation (opposed to `zunpkd8xy` insns).

There are also some lack of logic in shift-amount-operands' types.
E.g. `sra32[i][.u]` checks whether `sa` is positive, but the second argument of its intrinsic is `uint32_t`. It seems like -1 is much more adequate immediate value than 0x1F (even though -1 == 0x1F this case) so `int32_t` is more verbose.

marcfedorow, 2021-04-29:
It seems like `__rv__bitrev` should take `uint32_t`.
It is very inconvenient that there is no direct way to know `R[a:b]` is signed or unsigned.
E.g. `msb = Rs2[4:0];` in `bitrev` gives an unsigned value but `if (sa u> 0) {` implies that `sa` could be negative (so we need to specify that this is an unsigned comparison).

chuanhua, 2021-10-04:
For `kabsw`, since its result is still represented in a signed 32-bit range (i.e. saturated to 0x7fffffff), I think it is more appropriate to use int32_t as its return type.

chuanhua, 2021-10-04:
@marcfedorow Does the current spec still has problems relating to this issue?