riscv/riscv-p-spec #46: Typo fix

marcfedorow, 2021-04-29:
There are some markdown problems:
`smal[x]da`: common part `R[Idx1].R[Idx0] = R[Idx1].R[Idx0] + SE64(Mres0[31:0]) + SE64(Mres1[31:0]);` should be separated from x-specific code as in `smsl[x]da`. Same for rv64 version.
`kmda`, `kmatt`, `kmsxda32`, `uraddw`, `umul8` and probably some others have their code outside the `operations` box.
`srai8[.u]` have wrong comments, e.g. `//SRA8` instead of `//SRAI8`.
`kmatt` also has its intrinsic outside the `required` box.
`kmatt32` has its `exceptions` block inside the `operations` box.
`srl16` has its purpose in the `syntax` box.

Also there are some inconsistencies in `operations` blocks between different insns.
E.g. `km[x]da` insns check if any rs is **not** equal to something that cause a saturation but `kmmawb2` checks if both of them are equal to that value. There is no difference for a machine (even for simulator if branch predictor is good enough) but it could be more readable if only one option was provided.

Also it is unexpectedly to see `.W[0]` notation in short descriptions and `[31:0]` notation in `operations` blocks.

marcfedorow, 2021-04-29:
Also `Rd[63:0] = SE(resw[31:0])` or `Rd = SE64(Rs1.W[0])` could be replaced.
While SE with no value is not described in `2.1. Shorthand Definitions` section I think the latter is better.

chuanhua, 2021-05-03:
The above issues will be fixed in the next release. Thanks for finding these problems.

marcfedorow, 2021-06-23:
SRL32[.u] have SRA32[.u] comments in their code block.

chuanhua, 2021-06-23:
The above issue will be fixed in the next release. Thanks for finding this.

marcfedorow, 2021-06-25:
KMADA/KMAXDA operations block: missing indentation for `OV = 1;` line (negative saturation branch)

chuanhua, 2021-06-27:
Thanks. The above issue will be fixed in the next release.

chuanhua, 2021-10-12:
The above issues are fixed in ff37814.

closed 2021-10-12 by chuanhua