riscv/riscv-p-spec #50: The difference between p-spec and v-spec

Arima-Kawori, 2021-05-13:
We know that P-ext means "packed SIMD" and V-ext means "vector". I've read the docs of v-spec detailedly, and obviously v-ext also provide useful SIMD operations, and some operations are the same with those provided by P-ext. So I'm so curious why P-ext and V-ext are developing simultanously and what the differences between p and v-ext are.
Thanks for anyone who answers my questions!

linsinan1995, 2021-05-13:
Instructions from P extension operate on general-purpose registers instead of wider vector registers in V extension. This brings advantages, such as flexibility and power efficiency, and constrained devices can be enabled to process with high performance.

Arima-Kawori, 2021-05-15:
I get it. Thank you for your answer!

closed 2021-05-15 by Arima-Kawori