riscv/riscv-p-spec #51: KDMBB, KDMBT, KDMTT using V2HImode in RV64P

linsinan1995, 2021-05-18:
instructions, KDMBB, KDMBT, KDMTT, are only three instructions using V2HImode in RV64P. Perhaps it is a typo?

chuanhua, 2021-05-21:
RV64P `KDMAxx` instructions only use the first 32-bit part of the 64-bit registers as two 16-bit input operands. So the input operand type of `int16x2_t` in the RV64 vector-style intrinsic function is correct. For RV64 SIMD-type of similiar operations, you should use `KDMAxx16` instructions.

linsinan1995, 2021-06-09:
Does it seem that `kdmxx` also works in this way, but `kdmxx` are using v4hi for 64bit in vector mode.

linsinan1995, 2021-09-06:
I close this issue since kdmxx in rv64p are changed to v2hi in the commit 86876e75b8f56d2b26b0f813bde5f6ae07f028a6.

closed 2021-09-06 by linsinan1995