riscv/riscv-p-spec #52: Register pairs should always be little-endian order

jhauser-us, 2021-05-23:
The issue of how register pair order is affected by endianness came up with regards to the Zfinx extension (https://github.com/riscv/riscv-zfinx/issues/4).  To quote from that issue ticket:

> I've spoken with Andrew Waterman about this, and we agree that the P extension has it wrong and will need to change. Register pair order should *not* be affected by the current endianness. The main reason is that this would be very bad for RISC-V cores that support both little-endian and big-endian byte orders. Those cores would need muxes to select the low and high halves from each register operand pair ahead of operations. By fixing the register pair order to be only little-endian, those muxes can be eliminated.
>
> It is true that bi-endian cores might then be forced to swap the word order on loads and stores, if there are ever any instructions to load/store register pairs. But bi-endian cores already must swap byte order for regular loads and stores of halfwords, words, etc. And it would be far better to insert these muxes in the load/store path than in the ALU paths.
>
> Without load/store instructions of register pairs (a likely scenario), then it simply falls to software to load/store registers in the correct little-endian pair order, and no extra hardware muxes are needed either for the ALU paths or for memory accesses.

chuanhua, 2021-05-26:
Thanks for the feedback. I will change the big-endian register-pair order of P extension to be consistent with the ZDINX extension.

jnk0le, 2021-06-25:
I think it's already fixed in 0.9.5

closed 2021-06-26 by jhauser-us