riscv/riscv-p-spec #53: CLZ overlap

aswaterman, 2021-05-25:
The CLZ32 P instruction performs the same function as CLZ[W] in Zbb.

My recommendation is to remove the existing CLZ32 from P, then add CLZ[W] to P.  (It's OK for multiple extensions to include the same instruction.)

chuanhua, 2021-07-02:
CLZ32 does not overlap with CLZW in RV64. CLZ32 is a SIMD CLZ operation on two 32-bit elements in RV64. The overlap only happens in RV32. What is your suggestion for this condition?
- Replace CLZ32 with CLZ in RV32P and keep CLZ32 in RV64P and no CLZ in RV64P.

aswaterman, 2021-07-02:
My mistake, @chuanhua -- I missed that it's a SIMD operation, so it is only redundant for RV32.

chuanhua, 2021-09-15:
Release v0.9.7 (0560ad6) adds `clz` to replace `clz32` in RV32.

chuanhua, 2021-09-27:
Some usability issues are mentioned in #94. I think the comments there are worth considering for how to handle instruction overlaps if the overlaps only happen either for RV32 or RV64, but not both.