riscv/riscv-p-spec #54: MINW/MAXW overlap with MIN/MAX in Zbb

aswaterman, 2021-05-25:
MINW/MAXW perform the same function as MIN/MAX.

My recommendation is to remove MINW/MAXW from P, then add MIN/MAX to P.  (MIN/MAX would then belong to both P and Zbb.)

chuanhua, 2021-08-11:
MINW/MAXW only overlaps with MIN/MAX on RV32, not on RV64. Do you suggest that
* We add MIN/MAX to P, and remove MINW/MAXW on RV32.
* We keep MINW/MAXW, and also add MIN/MAX to P on RV64.

Most of RVP overlaps on RVB has this property that they only overlap either with RV32 or RV64, but not both. Do you have a general solution/idea for how to handling this kind of instruction overlaps?

aswaterman, 2021-08-11:
As a general recommendation, I would err on the side of avoiding redundant instruction encodings for a given base ISA: i.e., favor the first option.

(Of course, there might be specific cases where it makes sense to contravene that recommendation.)

chuanhua, 2021-09-15:
Release v0.9.7 (0560ad6) adds `max/min` to replace `maxw/minw`.