riscv/riscv-p-spec #55: BPICK overlap with CMIX in Zbt

aswaterman, 2021-05-25:
These instructions perform the same function.

Zbt is not yet frozen, so the resolution is somewhat arbitrary: we can delete either instruction, then add the other instruction to both P and Zbt.

jnk0le, 2021-05-25:
Zbt can be potentially placed into reserved FMA4 encoding riscv/riscv-bitmanip/issues/112

chuanhua, 2021-08-13:
I prefer to keep BPICK in P and add it to Zbt (i.e., delete CMIX in Zbt).

jnk0le, 2021-08-15:
Should the rest of Zbt land in OP-P as well?

chuanhua, 2021-08-17:
From the perspective of keeping all Zbt instructions together in a group, I would delete BPICK in P and add CMIX from Zbt to a P sub-extension, Zbpb, created to handle overlapping issues with B extension.

chuanhua, 2021-09-15:
Release v0.9.7 (0560ad6) adds `cmix` to replace `bpick`.