riscv/riscv-p-spec #56: PK*/PACK* overlap with Zbp

aswaterman, 2021-05-25:
PKBB16/PKBB32/PKTT16/PKTT32 overlap PACK[U][W] in Zbp.

Zbp is not yet frozen, but a subset _is_ frozen: Zbb includes `zext.h`, which maps to `pack[w] rd, rs1, x0`.

My recommendation is to retain the Zbp encodings for PACK[U][W].  We then delete the corresponding PKBB16/PKBB32/PKTT16/PKTT32 instructions from P, then add PACK[U][W] to P.

Finally, we should consider changing the encoding of PKBT16/PKBT32/PKTB16/PKTB32 to be more similar to the encoding of PACK[U][W].

chuanhua, 2021-09-15:
Release v0.9.7 (0560ad6) adds `pack` to replace `pkbb16` in RV32 and `pkbb32` in RV64.
Release v0.9.7 (0560ad6) adds `packu` to replace `pktt16` in RV32 and `pktt32` in RV64.

chuanhua, 2021-09-27:
Some usability issues are mentioned in #94. I think the comments there are worth considering for how to handle instruction overlaps if the overlaps only happen either for RV32 or RV64, but not both.