riscv/riscv-p-spec #57: WEXT[I] overlap with FSR[I][W] in Zbt

aswaterman, 2021-05-25:
WEXT[I] performs the same function as FSR[I] (RV32) or FSR[I]W (RV64).

Zbt is not yet frozen, and so the resolution is somewhat arbitrary.  But my recommendation is to delete WEXT[I] and then add FSR[I][W] to P, because the FSR* are generalized to 64-bit doublewords in RV64, whereas WEXT* is specific to 32-bit words.

jnk0le, 2021-06-07:
On RV64 `wext[i]` at most saves 1 base instruction. (right shift + zero extend)
If zero extension is not necessary (e.g. result consumed by instruction that ignores upper half) then it doesn't save anything.
It can also be replaced with `bext`, but that requires mask to be setup in one register (as a loop invariant).

On RV32 it's less flexible `fsr[i]`

jnk0le, 2021-06-11:
`bext` as per 0.93 bitmanip. rc1 uses that name for single bit extract.

chuanhua, 2021-09-06:
For RV32, `wext` functionality is a subset of `fsr`. For RV64, `wext` functionality is a subset of 
```
srai r3, rs1, 32
fsrw rd, rs1, rs3, rs2
```

marcfedorow, 2021-09-10:
So what will be done to wext?
Will it be removed?
Will be fsr and fsrw required by P extension?

chuanhua, 2021-09-15:
Release v0.9.7 (0560ad6) adds `fsr` and `fsrw` to replace `wext`.

chuanhua, 2021-09-27:
Some usability issues are mentioned in #94. I think the comments there are worth considering for how to handle instruction overlaps if the overlaps only happen either for RV32 or RV64, but not both.