riscv/riscv-p-spec #62: make regpair handling consistent with Zfinx

jnk0le, 2021-05-25:
In addition to #52

I think we should regularize regpair handling across risc-v extensions especially considering the fact that Zdinx has high chance to co-exist with P extension.

>Zfinx:
>When a double-width floating-point result is written to {\tt x0}, the entire
write takes no effect: e.g., for RV32Zdinx, writing a double-precision result
to {\tt x0} does not cause {\tt x1} to be written.

>Zfinx:
>When {\tt x0} is used as a double-width floating-point operand, the entire
operand is zero---i.e., {\tt x1} is not accessed.

P spec currently says
>since R0 is X0, i.e., 0, a constant, the pair of { R0, R1 } cannot be used in these instructions.

I interpret this as the useage of {R0,R1} pair is reserved.

>Zfinx:
>The specified register must be even, odd registers will cause an illegal instruction exception.

In P the lsb in specifier seem to be ignored (not very clean in the spec)
>Also, although the instruction encodings of these instructions allow a software developer to specify the pair number of an operand using either an even or an odd number of register index, an assembler or instruction geneator may choose to block the use of an odd number as a register index for the paired-register operand.

clarified more straightforward here
https://github.com/riscv/riscv-p-spec/issues/22#issuecomment-741669267
>For a register pair, only the pair group index [4:1] is meaningful. The register index[0] will be ignored and its value is a don't-care value.

jnk0le, 2021-05-25:
also

>8. x0 register target
If a floating point instruction targets x0 then it will still execute, and will set any required flags in fcsr. It will not write to a target register.

It should be the same for saturation flag in P

chuanhua, 2021-05-26:
Thanks for the comments. It is better for RISC-V register-pair scheme to be consistent across different extensions. I will make the register-pair of P extension to be consistent with the ZDINX extension.

jnk0le, 2021-06-25:
I think it's fixed in 0.9.5

closed 2021-06-25 by jnk0le