riscv/riscv-p-spec #65: KHMX8 bad operation destription

marcfedorow, 2021-06-17:
```
if (is “KHM8”) {
  op1t = Rs1.B[x+1]; op2t = Rs2.B[x+1]; // top
  op1b = Rs1.B[x];   op2b = Rs2.B[x];   // bottom
} else if (is “KHMX8”) {
  op1t = Rs1.H[x+1]; op2t = Rs2.H[x];   // Rs1 top
  op1b = Rs1.H[x];   op2b = Rs2.H[x+1]; // Rs1 bottom
}
```
I suppose RsX.B must stay instead of RsX.H

chuanhua, 2021-06-19:
Thanks for finding this issue. I will fix this typo in the next release.

marcfedorow, 2021-08-03:
`For the “KHMX16” instruction, multiply the top 8-bit Q7 content of 16-bit chunks in Rs1 with the bottom 8-bit Q7 content of 16-bit chunks in Rs2. At the same time, multiply the bottom 8-bit Q7 content of 16-bit chunks in Rs1 with the top 8-bit Q7 content of 16-bit chunks in Rs2.`
I believe there should stay `For the “KHMX8” instruction`.

chuanhua, 2021-08-04:
Thanks. I will fix it in the next release.

chuanhua, 2021-08-11:
Fixed in 7ca35508bf5.

closed 2021-08-11 by chuanhua