riscv/riscv-p-spec #68: Importance of accumulate insns?

marcfedorow, 2021-06-17:
KDMATT x1 x2 x3 (2 cycles) de facto is KDMTT x4 x2 x3 (1 cycle) + KADDW x1 x1 x4 (1 cycle).
Why do we need all the insns like this? Is it riscv approach?
Do we really need to save this 'forth register' while some other insns just use six of them?

chuanhua, 2021-06-23:
One reason to use KDMATT over "KDMTT + ADDW" is that if there is no data dependency between a sequence of KDMATT instructions, the throughput of KDMATT can be 1 per cycle, while the throughput of "KDMTT+ADDW" is 0.5. This is important for the performance of small kernel loop functions for DSP applications.

Another reason is that KDMABB can be used to implement the L_mac function which is one of the most useful multiply-accumulate operations for G.729, AMR, EVS, and other voice codecs.

KDMAxx instruction also has code size and instruction fetch bandwidth saving benefits over "KDMxx+ADDW".

marcfedorow, 2021-06-23:
My thought was based on a possibility to fetch two instructions simultaneously.
In a single-issue case that's a good point.
Thank you for your answer!

closed 2021-06-23 by marcfedorow