riscv/riscv-p-spec #69: bitrevi / srai.u "Format" block

marcfedorow, 2021-06-19:
`srai.u` has `imm6u`  fieldin bits 25:20.
`bitrevi` has `imm[5]` field in bit 25 and `imm[4:0]` field it bits 24:20.

There is no actual difference between these two insns (in terms of format) so IMO format block style should be similar in both cases.

Also it is not very clear from spec if `110101_1_00000_00001_001_00001_1110111` is legal insn on RV32 (if so, it should be recognized as `srai.u x1 x1 0`) or not (same for bitrevi).

Probably the code block should be rewrited in terms of XLEN, i.e.
```
sa = imm6u[log2(XLEN)-1:0];
if (sa > 0) {
  res[XLEN-1:-1] = SE65(Rs1[XLEN-1:(sa-1)]) + 1;
  Rd = res[XLEN-1:0];
} else {
  Rd = Rs1;
}
```

chuanhua, 2021-06-27:
`110101_1_00000_00001_001_00001_1110111` is not a legal insn on RV32. I will clarify the encoding format in the next release for both srai.u and bitrevi instructions. Thanks for pointing out this issue.

chuanhua, 2021-08-11:
Fixed in [7ca35508b](https://github.com/riscv/riscv-p-spec/commit/7ca35508bf5a618afd5658269e9918bfcdab705b).

closed 2021-08-11 by chuanhua