riscv/riscv-p-spec #94: confusion on ZBPBO subext

linsinan1995, 2021-09-16:
Hi,

I am a bit confused by how zbpbo works, so I open this issue. Take `clz` as example, 
> RV32: Replaced with CLZ in Zbpbo. RV64: Zpn

Does it mean, in rv32 we generate `clz` when the arch has zbpbo, and still can generate `clz32` when the arch does not have zbpbo?

if so, instructions in zbpbo can be covered by original p instructions, and what is the point of this change?

Thanks!

chuanhua, 2021-09-21:
For RV32, `clz` is in Zbpbo extension and Zpn does not have `clz32` instruction.
For RV64, Zpn has the `clz32` instruction.

I am sorry that the description in CLZ32 is not clear.

linsinan1995, 2021-09-22:
Thanks for replying. I was thinking that if clz32 already takes one encoding, why don't we just allow rv32 to have clz32?

Also, the instruction `rev` is a subset of `bitrev`/`bitrevi` and cannot completely replace `bitrev`/`bitrevi`, so it would be nicer and more clear if we can have more descriptions on it. E.g. as for `bitrevi`, the description "replaced with REV in Zbpbo extension when the immediate field is XLEN-1." instead of just "replaced with REV".

JamesKenneyImperas, 2021-09-23:
I agree that this is very confusing. Are instructions that are indicated as fully replaced by something in the Zbpbo extension (e.g. BITREV) ever implemented now? If they are not implemented, why not just delete them from this specification?

JamesKenneyImperas, 2021-09-24:
Am I alone in thinking the focus on eliminating overlap with instructions in the B extension is to the detriment of other important concerns (like usability)? This extension adds about 330 instructions, of which 14 have been identified as overlaps with the B extension. I have tried to summarize the situation in the list below, showing which original P extension instructions map to B extension replacements, on RV32 and RV64. To try to make this readable, I'm using lower case for P-extension instructions and **UPPER CASE BOLD** for B-extension replacements.

- clz32:          **CLZ** (RV32),         clz32 (RV64)
- pkbb16:      **PACK** (RV32),       pkbb16 (RV64)
- pktt16:        **PACKU** (RV32),    pktt16 (RV64)
- pkbb32:      n/a (RV32),                **PACK** (RV64)
- pktt32:        n/a (RV32),                **PACKU** (RV64)
- wexti:          **FSRI** (RV32),        wexti (RV64)
- wext:           **FSRI** (RV32),       **FSRW** (RV64)
- maxw:         **MAX**
- minw:          **MIN**
- swap8:        **REV8.H**
- bpick:          **CMIX**
- bitrev:         **REV**
- bitrevi:        **REV**

(It's possible I've made mistakes here - if so I think that reinforces the point about usability.)

I think it makes sense to for `bpick` to be mapped to `cmix`: this four-register instruction consumes a lot of encoding space. `maxw` and `minw` also look like sensible remappings.

`bitrev` and `bitrevi` only make sense if the replacement `rev` instruction exactly replicates them; the comment above makes me doubt this. There is no point replacing just a subset of an encoded instruction with something else.

Replacing `swap8` with `rev8.h` also looks fraught with problems. `rev8.h` is a subset of the overworked `grevi` instruction with shift amount 8. The Cryptographic (K) extension additionally requires variants of 'grevi' with shift amounts 7, 24, 31 (RV32 only), 56 (RV64 only) and 63 (RV64 only); here is another to add to that list. In a system in which B, K and P extensions are all present and separately enabled, it is really unclear to me which shift values should be allowed and which should cause Illegal Instruction traps for any particular combination of B, K and P enable.

For `pkbb16` and `pktt16`, I don't think it makes sense to replace them only on RV32. This saves no encoding space, since the instructions are still required for RV64, unless there is some plan to reuse that pattern for some entirely different purpose on RV32 only, which would be horrible. It also introduces a nasty inconsistency between RV32 and RV64, and also within RV32 itself (with `pktb16` and `pkbt16`).

The RV64-only `pkbb32` and `pktt32` replacements introduce the same kind of nasty inconsistency with  `pktb32` and `pkbt32`.

Replacements for `clz32` and `wexti` also make no sense because they are RV32-only, unless there is some plan to reuse that pattern for some entirely different purpose.

In conclusion, apart from `bpick`, `maxw` and `minw`, is there any real benefit to sharing B extension instructions? It saves a small fraction of the encoding space for this extension while introducing nasty usability issues.

linsinan1995, 2021-09-24:
I also put another question here,

Since instruction `wext` is its subset of `fsr` in terms of functionality, it seems that we need a new codegen for `fsr` in rv32. So `fsr` is only for rv32 in the spec, and rv64 do not have `fsr` instruction? It looks quite strange to me or did I misunderstand the spec?

chuanhua, 2021-09-27:
#59 has some discussions on using REV to replace bitrevi/bitrev.
The other overlaps are discussed in the following issues.
#53
#56
#57
#58

aswaterman, 2021-09-29:
We all have different objective functions, but I don't approve of adding redundant instructions across extensions.  Some of the griping here is purely aesthetic.  Re: `grevi`: "it is really unclear to me which shift values should be allowed and which should cause Illegal Instruction traps" -- if you were to pretend that the general `grevi` doesn't exist, and that the immediate field just happened to be an opcode field that distinguished rev8(RV32) from rev.b from rev8.h, you probably wouldn't be making the same comment.