riscv/riscv-p-spec #97: smax64 on rv64

marcfedorow, 2021-10-04:
Zpn 0.9.5. did not require `smax64`.
Zbpbo requires it tho via `max` (spec says "RV32 and RV64" instead of "RV32 only").
There is no information about adding new instruction required for P-extension.
Is this intended?
Same for rv64/min which could also be known as `smin64` but was not.

I do not think that it adds very much complexity whilest `radd64` requires the full 64-bit adder.
Anyway, this change should be described in a changelog.

chuanhua, 2021-10-05:
`max` in Zbpbo is replacing `maxw` in Zpn although `max` in RV64 is a super-set function of `maxw'. Similarly, `min` is replacing `minw'. Adding new instruction functions to RVP is not the intention. The intention is to resolve duplication issues with RVB.

I am sorry that I did not put more details in the revision history other than saying that some instructions in Zpn are replaced with instructions in Zbpbo.

marcfedorow, 2021-10-05:
riscv-k-spec requires rv32-only zip and rv32-only unzip.
Bitmanip spec v 0.9.3. describes both rv32 and rv64 zips (with a possible generalization to rv128).
This does not require other extensions to take the whole overlapping instruction.
If p-ext does not need rv64/max and rv64/min, just do not take them.
You can say that max and min are rv32-only in zbpbo.