riscv/riscv-p-spec #110: [The Q15 of Non-SIMD is it necessary to exist?]

kindleszzw, 2021-10-13:
The reasons are as follows:

1. About those(kaddh,ksubh,ukaddh,uksubh) instructions, I understand when and where they are used for。
   (kaddh description: The signed lower 32-bit content of Rs1 is added with the signed lower 32-bit content of Rs2. And the result is saturated to the 16-bit signed integer range of [-2^15, 2^15-1] and then sign-extended and written to Rd. If saturation happens, this instruction sets the OV flag.)

Because the result of Rs1+Rs2 may overflow on [RV32], If you don’t want it to overflow, you need 33-bit register support, and machines generally only support 32 and 64 bits, which makes it difficult for the compiler to automatically generate this instructions on [RV32] and similar to ukaddh.

And if “one kaddh16 = two kaddh”, then the kaddh instruction has no need to exist. Similar to ksub, ukaddh, uksubh. Please tell me the use cases for kaddh,ksub,ukaddh and uksubh。

2.There are other the Q15 of Non-SIMD instruction (khmbb, khmbt, khmtt), they can all use KHMX16 to replace and implement ,  so it’s a bit of a waste of instruction coding space.

kindleszzw, 2021-10-14:
Thank you for your replying.
So do you agree with my view? 
How about (u)k(add/sub)h those insts?

fioraking, 2021-10-18:
My boss told me some thing that i cant express it by english. But I understood it, so i close this issue.