riscv/riscv-p-spec #137: BitManip overlaps do not exist

allenjbaum, 2022-03-22:
Sec 4.2 lists Bitmanip instructions that do not exist (Zbt, Zpb)) as well as many opcodes that do not exist (FSR[I], CMIX, REV, REV8.H)

Section 5.1 lists these also.
These should either be moved to Zpn, or a new Bitmanip TG.

Kevin-Andes, 2022-03-28:
Hi Allen,

Thank you for the reminder. We are updating and revising the spec based on RVB version 1.0 instead of old 0.93. We will move the non-overlapping instructions back to Zpn so the actual overlapping instructions in Zbpbo may be reduced to just 3.

allenjbaum, 2022-04-05:
> This is actually a bad idea.
> For example, k-ext uses bitmanip overlaps on 0.9.3. insns such as xperms
> and brev8 (a.k.a. rev.b).
> When ZBT/ZBP will be ratified, similar P-ext insns will create an
> unnecessary overlaps.

This is complicated.
Zbkb is an improper subset of Zbb: it is missing some ops, contains the
rest ,and add some that have not been ratified
Zbkc is a       proper subset of Zbc: it is missing one op
Zbkx is completely composed of ops that we never ratified in the bitmanip
ratification package.
The intention was that bitmanip would come back later and add those - but
there is no signs that bitmanip is being reconstituted, which leaves these
in a really gray area.
I think I'll let Arch Review try to handwave this away...

jnk0le, 2022-04-05:
It should be handled the same way as crypto did Zbkb/Zbkx: Zbp

Zbt still needs allocation of major op to be handled by Arch Review
- pollute OP/OP-IMM
- FMA4
- OP-P

allenjbaum, 2022-04-06:
I am unaware of an activity to add more bitmanip extensions. That doesn't mean it hasn't been talked about, but there is no TG formed to look into it now, and it may be more than a little premature for P-extension to assume that they will happen.